tinyos/tinyos-main #328: tiny-os tools OS X no suitable python interpreter

ckalas, 2015-02-24:
This occurs when running ./configure in tinyos-main/tools. Any ideas why?

checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... config-aux/install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-apple-darwin14.1.0
checking host system type... x86_64-apple-darwin14.1.0
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for ranlib... ranlib
checking for a Python3 interpreter with version >= 3.0... none
checking for none version... ./configure: line 4557: -c: command not found

checking for none platform... ./configure: line 4576: -c: command not found

checking for none script directory... 
checking for none extension module directory... /lib/python/site-packages
checking whether : version is >= 2.5... yes
configure: error: no suitable Python interpreter found

ckalas, 2015-02-24:
Installing Python 3 seemed to have fixed that error, but which version of Python 2.x does it want?

ppannuto, 2015-02-24:
> checking whether : version is >= 2.5... yes

It looks like >= 2.5 should work.

ckalas, 2015-02-24:
The system python is 2.7.8 when I run python from terminal, this is why I'm unsure.

cire831, 2015-02-24:
2.7 should work.

the python code in the tinyos tree is python 2 code so any 2.x should work.

closed 2015-02-28 by ckalas