tinyos/tinyos-main #331: Disabling ext. PA on RFA1

che11, 2015-03-03:
Hello,
I've noticed that disabling ext. PA in RFA1 sw acks implementation (RFA1DriverLayerP.nc) differs from in hw acks implementation (RFA1DriverHwAckP.nc). 

in RFA1DriverLayerP.nc  ext pa will be disabled in state ( (cmd == CMD_TURNOFF || cmd == CMD_STANDBY) && state == STATE_RX_ON )

but in RFA1DriverHwAckP.nc only in ( cmd == CMD_TURNOFF && state == STATE_TRX_OFF )

Implementation from RFA1DriverLayerP seems to be more energy-efficient,or am I being wrong here?
Best Regards,
Elena Chervakova

andrasbiro, 2015-03-03:
Just to be clear, I think you mean the PA_EXT_EN bit in TRX_CTRL_1.

The problem here is this part of the datasheet:
"It is recommended to modify the transceiver registers in the address range 0x141 - 0x14D and 0x155 - 0x17F only, while in TRX_OFF state. In this case, no special procedures are required." (9.3.1.1, page 33)
And TRX_CTRL_1 is 0x144, so it should be modified only in TRX_OFF (or written twice which forces the update of a cached register). 

I overlooked this part when I wrote the swAck driver, but I tried to keep in mind when I ported the hwAck driver. I plan to merge these modifications into the swack driver as well.

By the way, it is strange that the swack driver never had any issue with this. Some of these registers doesn't even makes sense: TRX_STATE (0x142) controls the state machine of the radio, so you have to change it to reach the TRX_OFF state. I suspect there's something wrong with the datasheet, but I never had the time to dig into this.

che11, 2015-03-03:
ok, I referred to this part of the datasheet:
"It is recommended to set PA_EXT_EN=1 only in receive or transmit states to
reduce the power consumption or avoid leakage current of external RF switches or
other building blocks especially during SLEEP state."
You are right, in swack driver this command comes before TRX_OFF

andrasbiro, 2015-03-03:
And it doesn't really matter. FORCE_TRX_OFF should be 1us.

che11, 2015-03-03:
do we need to clear  PA_EXT_EN-bit also later, in CMD_STANDBY && STATE_TRX_OFF state?

che11, 2015-03-03:
I mean in hwack driver

andrasbiro, 2015-03-03:
Sorry, I lost track: I thought we talked about the PA_EXT_EN bit so far :) 
Which line is problematic? In which file?

che11, 2015-03-03:
in RFA1DriverHwAckP.nc you clear PA_EXT_EN-bit in case when cmd == CMD_TURNOFF and state == STATE_TRX_OFF (line 360), but not when cmd == CMD_STANDBY and state == STATE_TRX_OFF (line 370).

andrasbiro, 2015-03-03:
I see, you're right. It doesn't really matter, since standby state is not used by rfxlink (it is basicly the trx_off state of the chip), but yes, the PA should be turned off in standby state.
I will fix it when I'm in the correct branch.

che11, 2015-03-03:
ok thanks for your reply and explanations

che11, 2015-03-04:
another thing in RFA1DriverHwAckP.nc - line 338 and 426,427 - bit shifting seems to be wrong (direction), so that acks did not work on nodes with am addresses >255

andrasbiro, 2015-03-04:
Damn, you're right, I can't beleave I missed that :)

commit 71bf7d3f (2015-03-04): turn off the ext PA in standby mode (fixes #331)
closed 2015-03-04 by andrasbiro