tinyos/tinyos-main #334: sendDone error codes

Badisso, 2015-03-19:
Hi All,
It seems there is a mismatch in the error codes returned by the sendDone event (at least for TOSSIM). Thus, the AMSend and Send interfaces specify that sendDone returns FAIL if a transmission has failed (https://github.com/tinyos/tinyos-main/blob/master/tos/interfaces/AMSend.nc#L104). However, TossimPacketModelC.nc for example returns an EBUSY code for such a case (https://github.com/tinyos/tinyos-main/blob/master/tos/lib/tossim/TossimPacketModelC.nc#L245)
I am not sure if it is the case with other platforms, but a quick look to the CC2420TransmitP.nc (https://github.com/tinyos/tinyos-main/blob/master/tos/chips/cc2420/transmit/CC2420TransmitP.nc) seems to not return a FAIL code.
Please share your opinions on the appropriate fix

gnawali, 2015-03-19:
It is a bug in TOSSIM. EBUSY should be returned only when you call send when there is an outstanding send call, i.e., send was successful but sendDone is not signaled yet.

Badisso, 2015-03-19:
Yes, as you mentioned EBUSY is returned by the send command. The sendDone as documented in the AMSend and Send interfaces returns SUCCESS, FAIL or ECANCEL. In TOSSIM there might be a possibility for EOFF (https://github.com/tinyos/tinyos-main/blob/master/tos/lib/tossim/TossimPacketModelC.nc#L285). In the CC2420TransmitP it seems another code ERETRY can go up to both Send.sendDone and AMSend.sendDone (https://github.com/tinyos/tinyos-main/blob/master/tos/chips/cc2420/transmit/CC2420TransmitP.nc#L531)

gnawali, 2015-03-19:
I think we should update the documentation because EOFF and ERETRY are possible return values.

Badisso, 2015-03-19:
I agree about EOFF, but I think ERETRY should not be returned to upper layers which are more interested to know whether the transmission resulted in a SUCCESS or FAIL at the end.

gnawali, 2015-03-19:
many other places in the code use ERETRY so the question is in the radio stack -- what is the highest level to which it is appropriate to send ERETRY.