tinyos/tinyos-main #344: Tinyos 1.x download

xianlin, 2015-06-09:
Can I use 2.x to compile 1.x code? I think it's not compatible.

How to download/git clone TinyOS1.X? I have some old sensors need to use the old code. Thanks.

xianlin, 2015-06-09:
never mind and I have found the link:

How To Install TinyOS 1.x
http://leme.tagus.ist.utl.pt/gems/PmWiki/index.php/resources/HowToInstallTinyOS

cire831, 2015-06-09:
TinyOS 2.x is a different code base from TinyOS 1.x.  They are not compatible.  One doesn't use 2.x to compile anything.  Rather one combines ones code with modules from TinyOS 2.x then compile the result using an appropriate toolchain for ones h/w.

Sorry, I don't remember where one can get TinyOS 1 code.

closed 2015-06-09 by cire831