tinyos/tinyos-main #349: Ppprouter error when using make telosb

kentstudent, 2015-09-02:
Hi, 

I'm trying to build the ppprouter application using the "make telosb" command and getting the following error at the end of the output: 
collect2: ld returned 1 exit status
make: *** [exe] Error 1

I have searched on Github and other forums for this error but can't seem to find a solution. Any help greatly appreciated. Thanks

cire831, 2015-09-04:
can you please run this build again but with "verbose" in the make command.

then submit the entire output.

are you building off the current tip of the tinyos-main trunk?

kentstudent, 2015-09-04:
Hi,  
Thanks for the reply. Yes it is from https://github.com/tinyos/tinyos-main.git
Heres the output: 

GNU C (GCC) version 4.6.3 20120301 (mspgcc LTS 20120406 unpatched) (msp430)
	compiled by GNU C version 4.4.5, GMP version 4.3.2, MPFR version 3.0.0, MPC version 0.9
GGC heuristics: --param ggc-min-expand=98 --param ggc-min-heapsize=128460
ignoring nonexistent directory "/usr/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/sys-include"
ignoring duplicate directory "/usr/bin/../lib/gcc/../../lib/gcc/msp430/4.6.3/include"
ignoring duplicate directory "/usr/bin/../lib/gcc/../../lib/gcc/msp430/4.6.3/include-fixed"
ignoring nonexistent directory "/usr/bin/../lib/gcc/../../lib/gcc/msp430/4.6.3/../../../../msp430/sys-include"
ignoring duplicate directory "/usr/bin/../lib/gcc/../../lib/gcc/msp430/4.6.3/../../../../msp430/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/ncc
 ../../tos/lib/ppp
 ../../tos/lib/fragpool
 ../../tos/lib/net/
 ../../tos/lib/printf/
 ../../tools/tinyos/c/blip/
 ../../tos/lib/net/blip/
 ../../tos/lib/net/blip/interfaces/
 ../../tos/lib/net/blip/nwprog/
 ../../tos/lib/net/blip/shell/
 ../../tos/lib/net/blip/serial/
 ../../tos/lib/net/blip/platform/
 ../../tos/lib/net/blip/icmp/
 ../../tos/lib/net/blip/dhcp/
 ../../tos/lib/net/rpl
 ../../tos/platforms/telosb
 ../../tos/platforms/telosb/chips/stm25p
 ../../tos/platforms/telosa
 ../../tos/platforms/telosa/chips/cc2420
 ../../tos/platforms/telosa/chips/s1087
 ../../tos/platforms/telosa/chips/s10871
 ../../tos/platforms/telosa/chips/sht11
 ../../tos/platforms/epic/chips/ds2411
 ../../tos/chips/cc2420
 ../../tos/chips/cc2420/alarm
 ../../tos/chips/cc2420/control
 ../../tos/chips/cc2420/csma
 ../../tos/chips/cc2420/interfaces
 ../../tos/chips/cc2420/link
 ../../tos/chips/cc2420/lowpan
 ../../tos/chips/cc2420/lpl
 ../../tos/chips/cc2420/packet
 ../../tos/chips/cc2420/receive
 ../../tos/chips/cc2420/spi
 ../../tos/chips/cc2420/transmit
 ../../tos/chips/cc2420/unique
 ../../tos/chips/cc2420/security
 ../../tos/chips/msp430
 ../../tos/chips/msp430/adc12
 ../../tos/chips/msp430/dma
 ../../tos/chips/msp430/pins
 ../../tos/chips/msp430/timer
 ../../tos/chips/msp430/usart
 ../../tos/chips/msp430/sensors
 ../../tos/chips/stm25p
 ../../tos/chips/sht11
 ../../tos/chips/ds2411
 ../../tos/lib/timer
 ../../tos/lib/serial
 ../../tos/lib/power
 ../../tos/lib/onewire
 ../../tos/system
 ../../tos/types
 ../../tos/interfaces
 /usr/bin/../lib/gcc/msp430/4.6.3/include
 /usr/bin/../lib/gcc/msp430/4.6.3/include-fixed
[... 21 lines cut ...]
/tmp/ccZH8auj.o: In function `printf_in6addr':
app.c:(.text+0xca0): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/tmp/ccZH8auj.o: In function `RPLRankP__insertParent':
app.c:(.text+0xe30): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/tmp/ccZH8auj.o: In function `printf_ieee154addr':
app.c:(.text+0x1362): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
app.c:(.text+0x136e): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
app.c:(.text+0x1386): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
app.c:(.text+0x1398): undefined reference to `putchar'
/tmp/ccZH8auj.o: In function `IPDispatchP__lowpan_recon_start':
app.c:(.text+0x1e70): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/tmp/ccZH8auj.o: In function `IPDispatchP__reconstruct_age':
app.c:(.text+0x22a0): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
app.c:(.text+0x22d4): additional relocation overflows omitted from the output
/tmp/ccZH8auj.o: In function `IPNeighborDiscoveryP__IPForward__send':
app.c:(.text+0x2c32): undefined reference to `putchar'
/tmp/ccZH8auj.o: In function `IPProtocolsP__IP__send.isra.278':
app.c:(.text+0x4268): undefined reference to `putchar'
app.c:(.text+0x4316): undefined reference to `putchar'
/tmp/ccZH8auj.o: In function `IPForwardingEngineP__IPForward__recv.isra.327':
app.c:(.text+0x53ac): undefined reference to `putchar'
/tmp/ccZH8auj.o:app.c:(.text+0x62dc): more undefined references to `putchar' follow
collect2: ld returned 1 exit status
make: *** [exe] Error 1
root@tinyos:~/tinyos-main/apps/PppRouter#

vlahan, 2015-09-05:
PppRouter currently does not fit on the TelosB. As the error message says:

/usr/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: build/telosb/main.exe section .text' will not fit in regionrom'
/usr/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: section .vectors loaded at [0000ffe0,0000ffff] overlaps section .text loaded at [00004000,0001096f]
/usr/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region rom' overflowed by 5012 bytes

Take a look at the thread:

http://mail.millennium.berkeley.edu/pipermail/tinyos-help/2014-August/058544.html

There are several ways one can try to hack down the size, by removing different components of BLIP.

cire831, 2015-09-05:
it hasn't fit for awhile.  telosb is using the msp430f1611 which has if
memory serves 48KB of ROM.

I've been using the msp430f5438a which has more but still has problems
because of the tool chain.  Getting code into the upper > 64K rom space is
problematic.

kentstudent, 2015-09-08:
I initially tried to disable just the "components UDPShellC;" but still overflowed as below: 

/usr/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: build/telosb/main.exe section `.text' will not fit in region `rom'
/usr/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: section .vectors loaded at [0000ffe0,0000ffff] overlaps section .text loaded at [00004000,0001018f]
/usr/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `rom' overflowed by 2512 bytes

In the end disabled the following in the ppprouter.nc file as suggested in the link provided by @vlahan and the compile was successful. 
components RouteCmdC;
components UDPShellC;

[INFO] script
           45678 bytes in ROM
            9270 bytes in RAM
[INFO] size (toolchain):
   text	   data	    bss	    dec	    hex	filename
  45878	    132	   9140	  55150	   d76e	build/telosb/main.exe
[INFO] generating symbol table
[INFO] generating listing
[INFO] creating ihex file
[INFO] writing TOS image
[INFO] writing TOS buildinfo
[INFO] running the wiring check

For the time being this is ok for me. Thank you both for your help.

closed 2017-03-11 by cire831

navneetnivu07, 2017-03-24:
Hi, everyone! Even I'm getting the same error. What is the solution? Or is there any workaround?

cire831, 2017-03-24:
sorry the solution is to either wack what is being included or move to a more modern larger mote.

ppprouter hasn't fit for quite some time.

got back and read all the notes on this issue and you'll see what is going on.

navneetnivu07, 2017-03-25:
Thank you sir 👍  @cire831