tinyos/tinyos-main #351: TinyOS

rkertesz, 2015-09-07:
I work with a group that has used TinyOS for development for some time. I, however have been playing a lot with cortex processors such as the particle photon (STM32F205 ARM Cortex M3), a cortex m4 in the lesser known konekt dash (Freescale mk22fn512vlh12-nd), and even the arm cortex m0 in RFdigital's RFduino (really a Nordic chip 51822). I have been doing my product evaluation in wiring but I'd like to keep with TinyOS instead of something like mbed, wiring, etc. I saw that Tadashi Takaoka had worked on porting to cortex and see some header files here titled cortex but the effort may have gone stale. Is there any effort underway to bring tinyos to these processors?

cire831, 2015-09-07:
there is no active work that I know of on the cortex.

however the guys over at the University of Utah a couple of years ago had
the cortex m3 code working on the SAM evaluation module that worked.

They also did a TI cc2520 radio port (based on the cc2420 code) that I then
started with but ended up doing a rewrite as the cc2520-v2 code.

cire831, 2016-08-20:
Turns out we are moving to the TI MSP432 which includes a Cortex-M4F.  So I will be focusing on cleaning up the Cortex code.

Any interest in helping?

cire831, 2016-10-24:
Active work progresses...

cire831, 2017-07-23:
The Cortex-M4F is fully supported by code in the mainline of tinyos-prod.  See [https://github.com/tp-freeforall/prod].

Supports directly the TI msp432P401R (a Cortex-M4F core)

closed 2017-07-23 by cire831