tinyos/tinyos-main #356: TINYOS_ROOT_DIR_ADDITIONAL and TOSMake_include_make_platform

raidoz, 2015-10-11:
I'm trying to add some extras under an external tinyos-other tree(listed in `TINYOS_ROOT_DIR_ADDITIONAL`) and it does not seem to work, the original extras under the main tinyos tree become invisible.

Specifically, I'm trying to add a programmer type for AVR platforms, so I added:
`tinyos-other/support/make/avr/newporgrammer.extra`

Which I think should complement or override the stuff in:
`tinyos-main/support/make/avr/`

The avr stuff gets pulled in by the main target with `$(call TOSMake_include_make_platform,avr)`

Now when trying to install, the install/reinstall targets no longer work, the build system does not find them anymore. The newprogrammer is now listed as a valid extra, but install/reinstall and the programmers in the main tinyos are no longer listed (by `make iris help` for example).

A similar tinyos-other approach works just fine for adding *.target and *.platform files and even *.extra files, but not when adding things under a directory that is included with TOSMake_include_make_platform.

I can get around this by for example putting my stuff in `tinyos-other/support/make/avrplus`, adding an empty `avrplus.rules` and including `$(call TOSMake_include_make_platform,avrplus)`, but this requires me to also override existing AVR platforms to include that directory. I would like the programmer to be available for the default iris platform as well by simply including my tinyos-other in `TINYOS_ROOT_DIR_ADDITIONAL`.

So am I doing something that shouldn't work on principle, doing something wrong or is there a bug?

cire831, 2016-08-20:
did you make any progress on this issue?

raidoz, 2016-10-14:
I didn't, but if you agree that what I was trying to do should work and is the desired behavior, then I will dig a bit deeper into the make stuff.

cire831, 2016-10-14:
what you are trying to do is reasonable.