tinyos/tinyos-main #373: Add gcc >= 5.0 support

legobrick, 2016-05-29:
Right now make must be forced to use GCC=gcc-4.8 and GPP=g++-4.8.
Please add support for newer versions of gcc.

cire831, 2016-05-30:
This is an all volunteer project, so I would encourage you to make the effort and submit the changes.

closed 2016-05-30 by cire831

EduardDopler, 2016-05-31:
@legobrick I'm lost too, as my GCC 6 isn't supported. 
Could you please provide information how you make your build work? Where did you add the variables to point it to the old `gcc`?

legobrick, 2016-05-31:
@EduardDopler sure, not so used with GitHub, but I'll try to post the patch you need to apply!

First, you need to install GCC and G++ 4.8, or a compatible version of GCC/G++. Then you need to modify the references to gcc (or avr-gcc and so on) in support/make/ files as explained in the contained README.md. This in an example of my modification for enabling compilation for TOSSIM. By the way, it also includes a fix to the PYTHON_VERSION regex, since it was not working with mine (Python 2.7.11+).

```
diff --git a/support/make/extras/sim.extra b/support/make/extras/sim.extra
index 94210bd..634ceba 100644
--- a/support/make/extras/sim.extra
+++ b/support/make/extras/sim.extra
@@ -1,13 +1,13 @@
 #-*-Makefile-*- vim:syntax=make
 
-export GCC=gcc
-GPP=g++
+export GCC=gcc-4.8
+GPP=g++-4.8
 OPTFLAGS = -g -O0
 LIBS = -lm -lstdc++
 PFLAGS += -DTOSSIM -fnesc-nido-tosnodes=1000 -fnesc-simulate -fnesc-nido-motenumber='sim_node\(\)' -fnesc-gcc=$(GCC)
 PFLAGS += -I%T/lib/tossim
 WFLAGS = -Wno-nesc-data-race
-PYTHON_VERSION ?= $(shell python --version 2>&1 | sed 's/Python 2\.\([0-9]\)\.[0-9]+\{0,1\}/2.\1/')
+PYTHON_VERSION ?= $(shell python --version 2>&1 | sed 's/Python 2\.\([0-9]\)\.\([0-9+]\)*/2.\1/')
 
 TOSMAKE_BUILD_DIR = simbuild/$(PLATFORM)
 TOSSIMPY_DIR      = $(TINYOS_OS_DIR)/lib/tossim
```

EduardDopler, 2016-05-31:
Thanks @legobrick! I can report that gcc-4.9 works as well. I'm on Arch Linux. There, you have the caveat that you have to change `python` to `python2` in the regular expression to make the version checking work (because `python` defaults to the Python 3 binary). Also, it can be simplified a bit. The result is then:
```bash
PYTHON_VERSION ?= $(shell python2 --version 2>&1 | sed 's/Python 2\.\([0-9]\)\.[0-9]*/2.\1/')
```

SandipanBiswas777, 2017-12-03:
I have problem running command in terminal when using python in terminal for Tiny OS TOSSIM:-
from TOSSIM import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "TOSSIM.py", line 7, in <module>
    import _TOSSIM
ImportError: ./_TOSSIMmodule.so: undefined symbol: __nesc_atomic_end

please solve it.

cire831, 2017-12-05:
> I have problem running command in terminal when using python in terminal
> for Tiny OS TOSSIM:-
> from TOSSIM import *
> Traceback (most recent call last):
> File "", line 1, in
> File "TOSSIM.py", line 7, in
> import _TOSSIM
> ImportError: ./_TOSSIMmodule.so: undefined symbol: __nesc_atomic_end
>
> please solve it.

that isn't how this works.

this is an all volunteer project and folks working on the project all have
other things they do.

you should at least make an attempt at figuring out what has gone wrong and
fix it yourself.

and then contribute back to the project.

CentIIO, 2018-01-03:
gcc 4.8 fixed the  issue.Thank you @legobrick 
1) Install g++/gcc version 4.8 
2) Modify sim.extra to point to version in step 1 
           _export GCC=gcc-4.8_
             _GPP=g++-4.8_
3) No change in python regex (Active: 2.7.12)