tinyos/tinyos-main #376: tinyos-main/tools compile error

lucasmr, 2016-07-01:
`../../../../../tos/types/IeeeEui64.h:46:9: error: unknown type name ‘nx_struct’
 typedef nx_struct nx_ieee_eui64 {
         ^
../../../../../tos/types/IeeeEui64.h:46:33: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘{’ token
 typedef nx_struct nx_ieee_eui64 {
                                 ^
../../../../../tos/types/IeeeEui64.h:48:3: warning: data definition has no type or storage class
 } nx_ieee_eui64_t;
   ^
../../../../../tos/types/IeeeEui64.h:48:3: warning: type defaults to ‘int’ in declaration of ‘nx_ieee_eui64_t’ [-Wimplicit-int]
make[6]: *** [Makefile:422: lib6lowpan.o] Error 1`

Using gcc (Gentoo 5.4.0 p1.0, pie-0.6.5) 5.4.0 on amd64.

ocet, 2016-07-01:
I encountered same issue.

gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3)

supernlogn, 2016-07-03:
I just removed the definition of nx_ieee_eui64
in tos/types/IeeeEui64.h
and it compiled perfectly!

I even ran  the apps to see that all are ok.

alexbeltran, 2016-08-20:
Bug was introduced by pull request #374. nx_ieee_eui64 is not used anywhere else so it is safe to remove as suggested by @supernlogn .

closed 2016-08-20 by cire831

raidoz, 2016-08-24:
@supernlogn can you try your build with:

#ifdef NESC
typedef nx_struct nx_ieee_eui64 {
nx_uint8_t data[IEEE_EUI64_LENGTH];
} nx_ieee_eui64_t;
#endif // NESC

instead of completely removing nx_ieee_eui64

cire831, 2016-10-17:
NESC doesn't like this definition either.  Basically trying to define a low level primitive for Eui64 objects but the current definition doesn't work and NESC barfs.

I've looked around to see if I could see how other primitives are defined but no luck.

raidoz, 2016-10-17:
Where? What happens?

cire831, 2016-10-18:
using the tip of the tinyos-main(master):

apps/CoapPpp

In file included from /home/cire/mm/t2_cur/tinyos-2.x/tos/types/Ieee154.h:43:0,
                 from /home/cire/mm/t2_cur/tinyos-2.x/tools/tinyos/c/blip/lib6lowpan/lib6lowpan-includes.h:19,
                 from /home/cire/mm/t2_cur/tinyos-2.x/tools/tinyos/c/blip/lib6lowpan/lib6lowpan.h:29,
                 from /home/cire/mm/t2_cur/tinyos-2.x/tools/tinyos/c/blip/lib6lowpan/iovec.c:4:
/home/cire/mm/t2_cur/tinyos-2.x/tos/types/IeeeEui64.h:48:1: error: unknown type name ‘nx_struct’
/home/cire/mm/t2_cur/tinyos-2.x/tos/types/IeeeEui64.h:48:33: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘{’ token
/home/cire/mm/t2_cur/tinyos-2.x/tos/types/IeeeEui64.h:50:3: warning: data definition has no type or storage class [enabled by default]
/home/cire/mm/t2_cur/tinyos-2.x/tos/types/IeeeEui64.h:50:3: warning: type defaults to ‘int’ in declaration of ‘nx_ieee_eui64_t’ [-Wimplicit-int]

I don't know why.  I've tried various mods to the nx_ieee_eui64 struct and nothing made any difference.

what problem are you trying to solve...????

commit c3b23ce5 (2016-08-20): pull nx_struct nx_ieee_eui64

fixes #376