tinyos/tinyos-main #379: Load JNI failed on Raspberry Pi 3

Yaoshicn, 2016-09-22:
Hi, I am working on TinyOS recent days. When I try to use the MsgReader to read message from the mote, it failed to load JNI. The output is below:
`Error loading the TinyOS JNI libraries the conventional way!
Attempting to load library 'toscomm'
Could not load library 'toscomm': no toscomm in java.library.path
----------
java.lang.UnsatisfiedLinkError: no toscomm in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1864)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at net.tinyos.util.TOSLibraryLoader.load(TOSLibraryLoader.java:85)
	at net.tinyos.comm.TOSCommJNI.<clinit>(TOSCommJNI.java:14)
	at net.tinyos.comm.NativeSerial.getTOSCommMap(NativeSerial.java:207)
	at net.tinyos.comm.TOSSerial.<init>(TOSSerial.java:277)
	at net.tinyos.packet.SerialByteSource.openStreams(SerialByteSource.java:67)
	at net.tinyos.packet.StreamByteSource.open(StreamByteSource.java:61)
	at net.tinyos.packet.Packetizer.openSource(Packetizer.java:149)
	at net.tinyos.packet.AbstractSource.open(AbstractSource.java:78)
	at net.tinyos.packet.Packetizer.open(Packetizer.java:142)
	at net.tinyos.packet.PhoenixSource.run(PhoenixSource.java:178)
----------

In order to load the library 'toscomm' Java tries to locate the file 'libtoscomm.so' in one of the following paths:
/usr/java/packages/lib/arm:/lib:/usr/lib

The operating system is 'Linux' (arm)

The operating system and architecture is currently not supported
Exception in thread "Thread-1" java.lang.UnsatisfiedLinkError: net.tinyos.comm.TOSCommJNI.NativeSerial_getTOSCommMap()Ljava/lang/String;
	at net.tinyos.comm.TOSCommJNI.NativeSerial_getTOSCommMap(Native Method)
	at net.tinyos.comm.NativeSerial.getTOSCommMap(NativeSerial.java:207)
	at net.tinyos.comm.TOSSerial.<init>(TOSSerial.java:277)
	at net.tinyos.packet.SerialByteSource.openStreams(SerialByteSource.java:67)
	at net.tinyos.packet.StreamByteSource.open(StreamByteSource.java:61)
	at net.tinyos.packet.Packetizer.openSource(Packetizer.java:149)
	at net.tinyos.packet.AbstractSource.open(AbstractSource.java:78)
	at net.tinyos.packet.Packetizer.open(Packetizer.java:142)
	at net.tinyos.packet.PhoenixSource.run(PhoenixSource.java:178)
`

and I run the programme mentioned on http://mail.millennium.berkeley.edu/pipermail/tinyos-help/2010-June/046746.html&gws_rd=cr&ei=5OTjV6qdGIqq0gT5rrmICQ
`public class LibTest {
  public static void main(String[] args) {
    System.out.println(">> " + System.getProperty("java.library.path"));
    try {
      System.loadLibrary("getenv");
      System.loadLibrary("toscomm");
    } catch(Exception ex) {
      ex.printStackTrace();
    }
  }
`
The output is 
`>> /usr/java/packages/lib/arm:/lib:/usr/lib
Exception in thread "main" java.lang.UnsatisfiedLinkError: no getenv in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1864)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at LibTest.main(LibTest.java:5)
`
and my java version is 
`java version "1.8.0_65"
Java(TM) SE Runtime Environment (build 1.8.0_65-b17)
Java HotSpot(TM) Client VM (build 25.65-b01, mixed mode)
`
Would someone give me some help on this issue?
Thanks in advance.

Yaoshicn, 2016-09-23:
I hava solved this problem on the raspbian arm64 platform. Please kindly follow the steps below:
1. Try **sudo tos-install jni**
2. Check your jni path(tos-locate-jre --jni) and cp libgetenv.so to the jni path
3. Run LibTest again, you may find another error in LibTest which is toscomm not found
4. Check your java classpath and java **lib path** to see if libtoscomm.so is included
5. Add correct java classpath and lib path in bashrc, and then source bashrc
Done.

closed 2016-09-23 by Yaoshicn

cogle, 2017-04-22:
Could you list your java version with `update-java-alternatives -l` I am getting an error running `tos-install-jni` that states I do not have Java installed and I suspect it is due to differences in the Java version you have downloaded and the one I am using.