tinyos/tinyos-main #387: TinyOs and Zigbee protocol

Auron691, 2016-11-30:
Hi, i am new on the TelosB and TonyOs world, i have a simple question, it is possible to customize ZigBee protocol? for example modify the Network layer from a firmware?
thanks all for support

cire831, 2016-11-30:
hi Auron,

this isn't the place for questions like this.  you should use tinyos-help@millennium.berkeley.edu

ZigBee is just more software so yes you can change it?

but why?   what problem are you trying to solve?

closed 2016-11-30 by cire831

Auron691, 2016-11-30:
because i am studying thisprotocol with some device and i see that a specific packet, in this case Remote AT command, are parse from the device and don't send to Application layer, I cant read or decide a particular response. I want see if is possible to send this type of packet at Application layer

cire831, 2016-12-01:
you'll have to either try to follow the packet path through the code by reading it (painful) or set up a debugging rig and watch how the packet traverses the stack.

I doubt that anyone will respond but you might get lucky.