tinyos/tinyos-main #388: rf230: missing option for determining received signal power via PHY_RSSI when using hardware ACKs

miri-in, 2016-12-01:
If one uses the rf230 with software ACKs,
then there are __two possibilities__ in order to estimate
the received signal power, see RF230DriverLayerP.nc.

- If F230_RSSI_ENERGY is *not* defined, then
PacketRSSI is set to the value of register __PHY_RSSI__ (RSSI), see line 785;
otherwise, it is set to the value of register __PHY_ED_LEVEL__ (Energy Detection), see line 746.
- However, if one uses hardware ACKs, then there is no such choice and always Energy Detection, i.e., register PHY_ED_LEVEL, is used.

In contrast, for example both driver layers (software and hardware ACK) of atm128rfa1 support RSSI and Energy Detection.

cire831, 2016-12-02:
If one is using HW acks, I'm not sure if there is a way to get the RSSI out.  There needs to be a mechanism to indicate that a packet has been received and here is the RSSI value.  Not sure how that works when doing HW acks.

I'm not using the RF230, rather we are using the SiLabs si4468 radio.

miri-in, 2016-12-05:
As far as my limited understanding goes,
one can get an RSSI value any time in receive states,
like it is implemented in RFA1DriverHwAckP.nc line 701.
However, in extended operating mode (i.a. hardware ACKs) it is not recommended.
So, is there any deeper reason why the rf230 hardware ACK driver layer does not support 
RSSI, whereas it is supported by the rfa1 hardware ACK driver layer?

Thank you for your time!

cire831, 2016-12-05:
sorry, no idea.  its been too long since i looked at it.  have you taken a look at the datasheets?

miri-in, 2016-12-06:
I haven't read them in-depth yet, but perhaps this would be the right next step.