tinyos/tinyos-main #392: cant execute make for new platform

shyandsy, 2016-12-18:
#target
add genomote platform to new version of TinyOS.

#what i have done
from old version to new version
1. copy tos/platforms/genomote/    => tos/platforms/genomote/
2. copy support/make/genomote/    => support/make/genomote/
3. copy support/make/genomote.target   =>  support/make/target/genomote.target
4. created a empty file support/make/platforms/genomote.platform
5. installed the mas430-gcc, and the make telosb works fine

the content of genomote.target is
```
#-*-Makefile-*- vim:syntax=make
#$Id$

TARGET = genomote

#MSP_MCU = msp430x1611

# Disable MSP430 hardware multiply because it makes MSPGCC die
PFLAGS += -mdisable-hwmul
OPTFLAGS += -O

MSP_BSL ?= tos-bsl
MSP_BSL_FLAGS = --telosb

VOLUME_FILE = volumes-stm25p.xml
VOLUME_ALLOCATOR = tos-storage-stm25p

ifdef CC2420_CHANNEL
PFLAGS += -DCC2420_DEF_CHANNEL=$(CC2420_CHANNEL)
endif

# So that I know when I am compiling for genomote
CFLAGS += -DGENOMOTE

$(call TOSMake_include_platform,genomote)

ifeq ($(filter master slave,$(MAKECMDGOALS)),)
%:
	@echo "ERROR: Please specify whether you are compiling for master or slave"
else
genomote: $(BUILD_DEPS)
	@:
endif
```

the content of genomote.platform is
```
#-*-Makefile-*- vim:syntax=make
```

# error
when I execute the command    make genomote master
i got the following error message
```
nescc -o build/genomote/main.exe   -Os -O -gcc= -Wnesc-all -fnesc-include=tos -fnesc-scheduler=TinySchedulerC,TinySchedulerC.TaskBasic,TaskBasic,TaskBasic,runTask,postTask -fnesc-cfile=build/genomote/app.c -fnesc-separator=__ -mdisable-hwmul -DPLATFORM_GENOMOTE -I../../tos/system -I../../tos/types -I../../tos/interfaces -DIDENT_APPNAME=\"BlinkAppC\" -DIDENT_USERNAME=\"shyandsy\" -DIDENT_HOSTNAME=\"localhost.local\" -DIDENT_USERHASH=0x6c6cbabbL -DIDENT_TIMESTAMP=0x5856e078L -DIDENT_UIDHASH=0x32ed028cL -DGENOMOTE -fnesc-dump=wiring -fnesc-dump='interfaces(!abstract())' -fnesc-dump='referenced(interfacedefs, components)' -fnesc-dumpfile=build/genomote/wiring-check.xml BlinkAppC.nc   
Couldn't execute 
make: *** [exe] Error 2
```
[image]

cire831, 2016-12-19:
this is not the place for this.

please take this over to tinyos-help@millennium.berkeley.edu

closed 2016-12-19 by cire831

shyandsy, 2016-12-19:
@cire831  
according to my expriment, there is a mistake on the document for upgrade platform from 2 to 3

if you wanna fix that, i will show u

cire831, 2016-12-20:
yes please, email directly, cire831@gmail.com