tinyos/tinyos-main #396: stop of (last) virtual timer does not stop physical timer immediately

miri-in, 2017-03-01:
Consider an application where the last virtual timer is going to
fire at time t. When this timer is stopped at time t' < t, e.g., because of an
external event, then the underlying hardware timer remains active for time t - t'.
The reason is that in `VirtualizeTimerC.nc`, line 164, the command `Timer.stop`
only sets the` isrunning`-flag, but does not post  `updateFromTimer()`.

However, if the power management would be able to reach a better power state if
the hardware timer is inactive (e.g., in an application where the following wakeup is initiated via an asynchronous interrupt), it could be advantageous to post the update task from command `Timer.stop`.

cire831, 2017-03-01:
that is correct.  good find.  I'll fix it.

inouetoukyou, 2018-12-12:
has it been fixed?

cire831, 2018-12-14:
not that I am aware of.

i need to duplicate.

1st I will see if it shows up on the Msp432 Arm Cortex platforms.  As those
are the platforms I have with me.

If it reproduces there it will be quick for me to figure it out.

If not I will have to resurrect the telosb platform I have squirrled away.
That is why I haven't gotten to it.