tinyos/tinyos-main #402: Device type is not set when compiling for IRIS

jhnet, 2017-04-25:
Hi,
I tried to install tinyos from source on an Ubuntu 17.04. Compilation of TinyOS works fine. After installation I tried to compile the Blink App for a TelosB mote, which also works fine. 

When I try to compile for the iris mote, the avr/io.h complains about "device type not found". I installed gcc-avr and avr-libc from the official ubuntu repositories.
The complete error message is:
```
user@userVM:~/tinyos-main/apps/Blink$ make iris
[INFO] compiling BlinkAppC to a iris binary
nescc -o build/iris/main.exe   -Os -gcc=avr-gcc -Wnesc-all -fnesc-include=tos -fnesc-scheduler=TinySchedulerC,TinySchedulerC.TaskBasic,TaskBasic,TaskBasic,runTask,postTask -fnesc-cfile=build/iris/app.c -fnesc-separator=__ -I../../tos/platforms/iris -I../../tos/platforms/micaz -I../../tos/platforms/mica -I../../tos/platforms/iris/chips/rf230 -I../../tos/chips/rf230 -I../../tos/lib/rfxlink/layers -I../../tos/lib/rfxlink/util -I../../tos/platforms/iris/chips/at45db -I../../tos/platforms/mica2/chips/at45db -I../../tos/platforms/mica/chips/at45db -I../../tos/chips/at45db -I../../tos/platforms/iris/chips/ds2401 -I../../tos/platforms/mica2/chips/ds2401 -I../../tos/chips/ds2401 -I../../tos/chips/atm1281 -I../../tos/chips/atm1281/adc -I../../tos/chips/atm1281/timer -I../../tos/chips/atm128 -I../../tos/chips/atm128/adc -I../../tos/chips/atm128/pins -I../../tos/chips/atm128/spi -I../../tos/chips/atm128/i2c -I../../tos/chips/atm128/timer -I../../tos/lib/timer -I../../tos/lib/serial -I../../tos/lib/power -I../../tos/lib/diagmsg -mmcu=atmega1281 -fnesc-target=avr -fnesc-no-debug -DPLATFORM_IRIS -Wall -Wshadow --param max-inline-insns-single=100000 -Wno-unused-but-set-variable -Wno-enum-compare -I../../tos/system -I../../tos/types -I../../tos/interfaces -DIDENT_APPNAME=\"BlinkAppC\" -DIDENT_USERNAME=\"user\" -DIDENT_HOSTNAME=\"userVM\" -DIDENT_USERHASH=0xc400e479L -DIDENT_TIMESTAMP=0x58ff18f6L -DIDENT_UIDHASH=0x2090bb6eL -fnesc-dump=wiring -fnesc-dump='interfaces(!abstract())' -fnesc-dump='referenced(interfacedefs, components)' -fnesc-dumpfile=build/iris/wiring-check.xml BlinkAppC.nc -lm  
In file included from /usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:41,
                 from ../../tos/platforms/iris/platform.h:1,
                 from ../../tos/system/tos.h:43:
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/io.h:623:6: warning: #warning "device type not defined"
In file included from ../../tos/platforms/iris/platform.h:1,
                 from ../../tos/system/tos.h:43:
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h: In function `wdt_enable':
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:406: `WDT' undeclared (first use in this function)
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:406: (Each undeclared identifier is reported only once
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:406: for each function it appears in.)
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:417: `WDCE' undeclared (first use in this function)
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:417: `WDE' undeclared (first use in this function)
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h: In function `wdt_disable':
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:446: `WDT' undeclared (first use in this function)
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:460: `WDCE' undeclared (first use in this function)
/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/wdt.h:460: `WDE' undeclared (first use in this function)
In file included from ../../tos/system/TinySchedulerC.nc:52:
In component `McuSleepC':
../../tos/chips/atm1281/McuSleepC.nc: At top level:
../../tos/chips/atm1281/McuSleepC.nc:99: `SM0' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc:100: `SM2' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc:100: `SM1' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc:100: `SM0' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc:101: `SM1' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc:101: `SM0' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc:102: `SM2' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc:102: `SM1' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc:103: `SM1' undeclared here (not in a function)
../../tos/chips/atm1281/McuSleepC.nc: In function `getPowerState':
../../tos/chips/atm1281/McuSleepC.nc:115: `TIMSK0' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:115: `OCIE0A' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:115: `OCIE0B' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:116: `TIMSK1' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:116: `ICIE1' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:116: `OCIE1A' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:116: `OCIE1B' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:116: `OCIE1C' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:117: `TIMSK3' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:117: `ICIE3' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:117: `OCIE3A' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:117: `OCIE3B' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:117: `OCIE3C' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:122: `SPCR' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:122: `SPIE' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:126: `UCSR0B' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:126: `TXEN0' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:126: `RXEN0' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:129: `UCSR1B' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:129: `TXEN1' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:129: `RXEN1' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:133: `TWCR' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:133: `TWEN' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:137: `ADCSRA' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:137: `ADEN' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc: In function `McuSleep.sleep':
../../tos/chips/atm1281/McuSleepC.nc:149: `SMCR' undeclared (first use in this function)
../../tos/chips/atm1281/McuSleepC.nc:150: `SE' undeclared (first use in this function)
In file included from ../../tos/platforms/micaz/MotePlatformC.nc:44,
                 from ../../tos/platforms/mica/PlatformC.nc:56,
                 from ../../tos/system/MainC.nc:61,
                 from BlinkAppC.nc:56:
[... 284 lines cut ...]
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc: In function `TimerAsync.getAssr':
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:259: `ASSR' undeclared (first use in this function)
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc: In function `TimerAsync.setAssr':
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:263: `ASSR' undeclared (first use in this function)
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc: In function `TimerAsync.setTimer2Asynchronous':
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:267: `ASSR' undeclared (first use in this function)
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:267: `AS2' undeclared (first use in this function)
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc: In function `TimerAsync.controlABusy':
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:271: `ASSR' undeclared (first use in this function)
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:271: `TCR2AUB' undeclared (first use in this function)
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc: In function `TimerAsync.controlBBusy':
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:275: `ASSR' undeclared (first use in this function)
[... cut ...]
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:283: `OCR2BUB' undeclared (first use in this function)
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc: In function `TimerAsync.countBusy':
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:287: `ASSR' undeclared (first use in this function)
../../tos/chips/atm1281/timer/HplAtm1281Timer2AsyncP.nc:287: `TCN2UB' undeclared (first use in this function)
../../support/make/Makerules:369: die Regel für Ziel „exe“ scheiterte
make: *** [exe] Fehler 1
```

cire831, 2017-04-26:
> Hi,
> I tried to install tinyos from source on an Ubuntu 17.04. Compilation of TinyOS works fine. After installation I tried to compile the Blink App for a TelosB mote, which also works fine.
>
> When I [...]

Different motes/platforms use different toolchains.  The Iris toolchain uses the avr toolchain.  Something changed in the toolchain installed via Ubuntu 17.04 so it no longer recognizes the device type used on the IRIS.

I don't have the installation to figure this out.  So I'd like to encourage you to do the research to figure this out.

PraveenkumarDonga, 2017-06-09:
Hi!
I am also getting the same issue in ubuntu 16.04.
I am using this command to install the avr-toolchain.
sudo apt-get install gcc-avr binutils-avr gdb-avr avr-libc avrdude

Is there any update on this issue.
Looking forward for your help.

Thanks

cire831, 2017-06-09:
there is no update.

as I've said someone needs to figure out what the avr toolchain needs to recognize the chip being compiled for.

the iris platform calls out the atmega1281 cpu via -mmcu=atmega1281 in support/make/platforms/iris.platform

I don't know what the command line sequence for the new avr compiler looks like.  You will have to do the research into that.

You can see exactly what is being fed to the toolchain by doing "make iris verbose"

I have neither the time nor the set up to look into this.  Someone who is actively building for the atmegas will have to do the research into the toolchain.

cire831, 2017-06-09:
I did a quick google search for "avr-gcc command line options".  Got to this page [https://gcc.gnu.org/onlinedocs/gcc/AVR-Options.html] and then searched for "atmega1281" and found avr51.

So change the -mmcu=atmega1281 to -mmcu=avr51 in iris.platform.

that might work.

jhnet, 2017-06-09:
Actually version of avr-gcc I installed still accepts -mmcu=atmega1281 instead of -mmcu=avr51.
Verified this by compiling non-Tinyos avr blinking program. atmega1281 options compiles fine, avr51 not.

cire831, 2017-06-11:
well that would have been easy.

You'll have to dig in there and figure out what this line is bitching about:

/usr/lib/gcc/avr/4.9.2/../../../avr/include/avr/io.h:623:6: warning: #warning "device type not defined"

jhnet, 2017-06-12:
The warning is thrown, if no AVR has been defined. This is the corresponding section:
```
#else
#  if !defined(__COMPILING_AVR_LIBC__)
#    warning "device type not defined"
#  endif
#endif
```
It comes after many ifdefs to check which AVR has been defined. I uploaded the complete file. I thought that one of these devices should be defined by setting the -mmcu.
[io.txt](https://github.com/tinyos/tinyos-main/files/1067743/io.txt)

cire831, 2017-06-12:
it isn't matching any of the __AVR_<cpu_id>__ if defs.

the -mmcu is supposed to set something (I don't know the linkage) that then defines the __AVR_<cpu_id>__ bits.

but I see from above that you got a non-TinyOS program working with -mmcu=atmega1281  that is good to know.

please upload the command line you used to compile the non-TinyOS avr blink program

and upload the output of "make iris verbose"

also what source are you compiling from?

this sounds like something got wedged in the tinyos make system.

jhnet, 2017-06-13:
The command for compiling the non-TinyOs blink program is:
`avr-gcc -g -Os -mmcu=atmega1281 test.c`

The complete log of
`make iris verbose`
can be found here:
https://gist.github.com/jhnet/5b3441a83c6fe5d7016829fa4e28b073

cire831, 2017-06-14:
the "make iris verbose" is  using -mmcu=avr51

can you rerun it and fix -mmcu to be atmega1281?

should be in support/make/platforms/iris.platform.

jhnet, 2017-06-14:
Sorry, forgot to change this back to atmega1281 fom the test with avr51.
Here is the actual log
https://gist.github.com/jhnet/6c8d08ccefb102c54a89cceedc99c126

jhnet, 2017-06-14:
It seems that the avr51 is still used by some commands, although atmega1281 is set in iris.platform.

cire831, 2017-06-14:
how did you install the avr toolchain?

and what source are you using?

jhnet, 2017-08-10:
Sorry for the very late reply, I missed the notification..
AVR toolchain was installed from the official repositories, tinyos source is the one cloned from github.