tinyos/tinyos-main #405: missing default handler in Atm128AlarmC

miri-in, 2017-05-08:
The module `tos/chips/atm128/timer/Atm128AlarmC.nc` provides an interface Alarm with attribute @atmostonce(). Since this allows to omit wiring this interface, I would expect some default handler, like `default async event void Alarm.fired() {}`, in order to avoid possible "Alarm.fired not connected" errors.

cire831, 2017-05-08:
If one pulls in Atm128AlarmC it is because one wants the Alarm via Alarm.fired.  The purpose of atmostonce is to avoid fan out if someone erroneously does multiple wiring to Atm128AlarmC.  (its generic so has to be the same instance).

Getting the Alarm.fired not connected is actually a feature because if one pulls in Atm128AlarmC and doesn't connect Alarm.fired, I'd assert that is a bug and having the error thrown will clue one in that one forgot to do something.

closed 2017-05-08 by cire831

miri-in, 2017-05-09:
Hopefully, I am not wasting your time.

1. If one wants the interface to be wired, why not using @exactlyonce instead of @atmostonce?
2. Why allowing a default handler for  
  tinyos-main/tos/chips/atm128rfa1/timer/AtmegaAlarmP.nc
but not for 
  tinyos-main/tos/chips/atm128/timer/Atm128AlarmC.nc

cire831, 2017-05-09:
i don't know why Miklos added the default.  I wouldn't have done it.  there is a certain lack of consistency and discipline in the tinyos code.  It was kind of a free for all.

I would remove the default from atm128rfa1 but if it aint broke don't fix it.

I suspect everyone wiring to AtmegaAlarmP is also providing a .fired event.  but it isn't worth the time to go find out.

hopefully that gives you a little more perspective.

I punted your request because in my opininon it would make the code base just a tiny bit worse.  hopefully that is understandable.

also not wasting my time.  thanks for paying attention.  its appreciated.