tinyos/tinyos-main #408: Not able to program telosb after new installation

jobishjohn, 2017-06-18:
Hi all,

I recently installed tinyos 2.1.2 on ubuntu 14.04 (64 bit) from (deb http://tinyos.stanford.edu/tinyos/dists/ubuntu/ lucid main). But when I try to program the Blink application on a telosb , it gives me an error as shown below. Any idea what might be the issue? (The mote-hardware is ok as it works fine with an earlier installation in a different PC). Thanks in advance for your valuable time.

mkdir -p build/telosb
    compiling BlinkAppC to a telosb binary

    WARNING: Minimum recommended msp430-gcc version for this TinyOS release is 4.6.3!!!

ncc -o build/telosb/main.exe  -Os -fnesc-separator=__ -Wall -Wshadow -Wnesc-all -target=telosb -fnesc-cfile=build/telosb/app.c -board= -DDEFINED_TOS_AM_GROUP=0x22 -DIDENT_APPNAME=\"BlinkAppC\" -DIDENT_USERNAME=\"root\" -DIDENT_HOSTNAME=\"isllab-Vostro-3\" -DIDENT_USERHASH=0x1d7ed53eL -DIDENT_TIMESTAMP=0x5946dfe6L -DIDENT_UIDHASH=0x01d5f5c5L  BlinkAppC.nc -lm 
    compiled BlinkAppC to build/telosb/main.exe
            2528 bytes in ROM
              56 bytes in RAM
msp430-objcopy --output-target=ihex build/telosb/main.exe build/telosb/main.ihex
    writing TOS image
cp build/telosb/main.ihex build/telosb/main.ihex.out
    found mote on /dev/ttyUSB0 (using bsl,auto)
    installing telosb binary using bsl
tos-bsl --telosb -c /dev/ttyUSB0 -r -e -I -p build/telosb/main.ihex.out
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 1.61 (Device ID: f16c)
Changing baudrate to 38400 ...
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 1.61 (Device ID: f16c)
Changing baudrate to 38400 ...
Traceback (most recent call last):
  File "/usr/bin/tos-bsl", line 1918, in <module>
    main(0);
  File "/usr/bin/tos-bsl", line 1843, in main
    speed=speed,
  File "/usr/bin/tos-bsl", line 1218, in actionStartBSL
    self.actionChangeBaudrate(speed)            #change baudrate
  File "/usr/bin/tos-bsl", line 1345, in actionChangeBaudrate
    self.serialport.setBaudrate(baudrate)
AttributeError: 'Serial' object has no attribute 'setBaudrate'
/home/isllab/tinyos-main/support/make/msp/bsl.extra:45: recipe for target 'program' failed
make: *** [program] Error 1

cire831, 2017-06-18:
I don't know about the programming.  I don't work on any of the old TI based motes.  I'm currently using a modern ARM based mote from TI, the msp432p401r.

But the toolchain you are using from Stanford is outdated.  Follow the instructions from

http://tinyprod.net/repos/debian/

The msp430-gcc version there is 4.6.3

I don't know if that will fix your programming problem.

closed 2017-06-18 by cire831

JunhuaiYang, 2018-04-25:
I have this error too, so do you fix it?

vlahan, 2018-04-25:
@JunhuaiYang The error has nothing to do with the toolchain, but with a change in the pyserial API. Make sure that you are using the latest version of tinyos from github, in particular you should have the commit https://github.com/tinyos/tinyos-main/commit/7d54a311ecd7db685fc1aaeaa9473e50c08a1bfb included.

cire831, 2018-04-26:
> @vlahan <https://github.com/vlahan> Thank you very much. I have solved
> this problem. It is pySerial version problem.
> I use pip install "pySerial>=2.0,<=2.9999" . It works.

can you document this and submit a commit please.

vlahan, 2018-04-26:
@cire831 there is nothing to fix in the tinyos code. The fix is included in https://github.com/tinyos/tinyos-main/commit/7d54a311ecd7db685fc1aaeaa9473e50c08a1bfb. The fact that it works for @JunhuaiYang when he forces install of a 2.x pyserial version, indicates to me again that he is not using the latest tinyos pull but some old packaged version that does not include https://github.com/tinyos/tinyos-main/commit/7d54a311ecd7db685fc1aaeaa9473e50c08a1bfb