tinyos/tinyos-main #411: The fastserial implementation of UartStream.send in Msp430UartP only works with a buffer of 1 byte

MBradbury, 2017-06-28:
In fastserial/Msp430UartP.nc the UartStream.send function only works when the buffer has a length of 1 byte. This is a regression compared to chips/msp430/usart/MSP430UartP.n where UartStream.send supports a buffer of arbitrary length.

It looks like much of the chips/msp430/usart/MSP430UartP.nc could simply be used in fastserial/Msp430UartP.nc instead.

cire831, 2017-06-28:
if I fix it, do you have a way of testing it?

MBradbury, 2017-06-28:
I noticed this while trying to debug another issue. I'll get one of my TelosB motes set up and get it tested.

Yes this is on dev trunk.

cire831, 2017-06-29:
cool.  thanks.

i'm using arm processors solely at this point.   msp432.  good stuff