tinyos/tinyos-main #424: which version is this?

didiyou, 2018-07-05:


cire831, 2018-07-05:
what in particular are you talking about?

version of what?

didiyou, 2018-07-06:
en...I mean the Tinyos

cire831, 2018-07-06:
still makes not much sense.

If you are referring to the tip of tinyos main it is a development repository.  The tip of tinyos-main is currently... https://github.com/tinyos/tinyos-main/commit/ad056121cc51c449d6de154d98604e2c167dbfd8

which is the tip of the master branch in the repository.  It is well beyond the last official release.

tinyos-main is actually deprecated and new work is happening on the tp-freeforall repository.  See   https://github.com/tp-freeforall/prod

closed 2018-07-06 by cire831

didiyou, 2018-07-06:
OK，thanks，recently I always failed in installing  the Tinyos-2.1.2 in Ubuntu14.0.4,hence I want to find a recent version for study if it works,.Your help is greatly appreciated

anitaSpatil, 2019-01-22:
> repository. The tip of tinyos-main is currently... ad05612

Sir,
I want to work for TinyOS scheduling techniques,
 i.e I want to see  ,

   1. If  we can  implement other scheduling techniques in place of the
   default FIFO scheduler.
   2. If OS footprint ( base code of OS and executable file of application
   like Blink)  is occupying only around 400 bytes of  the memory in telosb
   like boards, then  before compiling and deploying the application on sensor
   board why can't we have multiple types of scheduling choices (for more
   efficiency with less waiting  and turn around times) instead of only FIFO?

So,for my work I have downloaded tinyos
 form https://github.com/tinyos/tinyos-main.
Sir,  should I continue with this or  download the tp-freeforall repository
i.e  https://github.com/tp-freeforall/prod in which  new work is happening.

Thank you
Sir.

cire831, 2019-01-22:
> Sir,
> I want to work for TinyOS scheduling techniques,
> i.e I want to see ,
>
> 1. If we can implement other scheduling techniques in place of the
> default FIFO scheduler.

TinyOS is built to allow this.  Look in tos/system/SchedulerBasicP.nc.

2. If OS footprint ( base code of OS and executable file of application
> like Blink) is occupying only around 400 bytes of the memory in telosb
> like boards, then before compiling and deploying the application on sensor
> board why can't we have multiple types of scheduling choices (for more
> efficiency with less waiting and turn around times) instead of only FIFO?

agreed.

So,for my work I have downloaded tinyos
> form https://github.com/tinyos/tinyos-main.
> Sir, should I continue with this or download the tp-freeforall repository
> i.e https://github.com/tp-freeforall/prod in which new work is happening.

tp-freeforall/prod would be better.

> Thank you
> Sir.
>
>
> >
> > If you are referring to the tip of tinyos main it is a development
> > repository. The tip of tinyos-main is currently... ad05612
> > <
> https://github.com/tinyos/tinyos-main/commit/ad056121cc51c449d6de154d98604e2c167dbfd8
> >
> >
> > which is the tip of the master branch in the repository. It is well
> beyond
> > the last official release.
> >
> > tinyos-main is actually deprecated and new work is happening on the
> > tp-freeforall repository. See https://github.com/tp-freeforall/prod
> >
> > <https://github.com/tinyos/tinyos-main/issues/424#issuecomment-402912533
> >,
> > <