tinyos/tinyos-main #428: How about TinyOS now？

yulincoder, 2018-12-01:
Actually, TinyOS is an excellent embedded OS either concept and implementation in my view and I love it. But I have been noticing that the community activity tends to decrease, How about TinyOS now in academia, industry and community now?
Thanks.

anitaSpatil, 2018-12-01:
Yes, I agree. I am the researcher in TinyOS and searching for the people working on TinyOS. 
Thanks.

cire831, 2018-12-03:
Hi,

TinyOS and nesC is harder to learn then more common approaches.  But I
agree that TinyOS and nesC are very well put together.

I am actively using TinyOS and nesC in developing embedded systems for
production quality products.

This work is being done over at github://tinyprod/prod and in particular
tp-freeforall/prod.    The majority of the work is focused on ARM
Cortex-m4F based platforms.  It is a superset of tinyos/tinyos-main.

let's work together.

eric

anitaSpatil, 2018-12-04:
Thanks a lot for your reply Sir Eric.
 I was eagerly waiting for reply. Actually I am the beginner in the area of TinyOS. I have knowledge about Operating Systems, Computer Networking and C programming language. I taught these subjects to UG students many times. With lot of interest I have taken TinyOS as my research area. I have many more confusions in programming using nesC , like, being an extended language of C does this supports all the features of C ?  etc. can you help me in this regard?

Let's work together....
Thanking you
Anita Patil

cire831, 2018-12-07:
> Thanks a lot for your reply Sir Eric.
> I was eagerly waiting for reply. Actually I am the beginner in the area of
> TinyOS. I have knowledge about Operating Systems, Computer Networking and C [...]

yes nesC is a superset of C.   Typically one trys not to get too clever in
C because that then violates the partitioning that nesC brings to the table.

What in particular are you struggling with?

Let's work together....
> Thanking you
> Anita Patil

yulincoder, 2018-12-07:
@cire831 hello, There is a PR  [#426](https://github.com/tinyos/tinyos-main/pull/426) I posted, Do you have any solution? 
@cire831 @anitaSpatil In addition, I have been making a tiny event-driving framework  [eframe](https://github.com/yulincoder/eframe) imitating the TinyOS.  hahaha 😄 😄 😄 
Thanks

cire831, 2018-12-09:
I've looked at it but haven't been able to figure out where it is going wrong.  I'll look at it again today or tomarrow.

anitaSpatil, 2018-12-11:
Sir,
 I have installed TinyOS-2.1.2 on Ubuntu 14.04 LTS and executed Blink application on simulator(I hope TOSSIM is the default simulator in tinyos-2.1.2 ) with
 1.  micaz  platforms using command
         avrora -simulation=sensor-network -seconds=5.0 -nodecount=5 main.elf
 2.  telosb platform using the command
         java -jar /opt/mspsim-master/mspsim.jar main.exe

 got the results also as it was shown in the referred youtube videos.

Now I would like to write some simple programs to see how printf ,scanf etc functions works in nesC. So, I referred the link
 http://tinyos.stanford.edu/tinyos-wiki/index.php/The_TinyOS_printf_Library
but it gives the example with hardware not the simulator. 

May I include C++ features like inline functions , virtual functions etc in tinyOS(I have seen c,cpp,java and python in tinyos-release-tinyos-2_1_2.zip\tinyos-release-tinyos-2_1_2\support\sdk )  ?

Actually I have to work on process scheduling in tinyOS. I need to study complete nesC in detail. 
I am referring the youtube videos and some tutorials to learn about tinyOS and nesC.
Sir, is this the right way of learning?
 Sir, as you said, tinyOS and nesC combination is harder, is there any other language for TinyOS?

Let's work together....

Thank you.

yulincoder, 2018-12-12:
There is a good tutorial [TinyOS programming](http://csl.stanford.edu/~pal/pubs/tos-programming-web.pdf) recommended you.

  
 
  
 
 http://tinyos.stanford.edu/tinyos-wiki/index.php/The_TinyOS_printf_Library

cire831, 2018-12-14:
> Sir,
> I have installed TinyOS-2.1.2 on Ubuntu 14.04 LTS and executed Blink application on simulator(I hope TOSSIM is the default simulator in tinyos-2.1.2 ) with
>
>     micaz platforms using [...]

It is my understanding that the simulator only works with micaz platforms.

> Now I would like to write some simple programs to see how printf ,scanf etc functions works in nesC. So, I referred the link
> http://tinyos.stanford.edu/tinyos-wiki/index.php/The_TinyOS_printf_Library
> but it gives the example with hardware not the simulator.

I don't know what that means.

There are various printf implementations.  It has evolved over time and was never consolidated with a single implementation.  I beleive there is a printf implmentation that uses low level uart h/w and a printf implmentation that is based on Active Messaging that works over the network or an AM serial connection.  I agree it is confusing.

> May I include C++ features like inline functions , virtual functions etc in tinyOS(I have seen c,cpp,java and python in tinyos-release-tinyos-2_1_2.zip\tinyos-release-tinyos-2_1_2\support\sdk ) ?

While technically you could do this, it is probably very difficult.  nesc is a preprocessor (it effectively replaces cpp) that feeds its output to gcc.  Not g++ the gnu C++ compiler.  GCC supports inline functions.

The other instances of code you are referencing are support tools and not used when building code for the underlying motes (the actual embedded system).

> Actually I have to work on process scheduling in tinyOS. I need to study complete nesC in detail.

Technically nesC is the implementation language used by tinyOS.

> I am referring the youtube videos and some tutorials to learn about tinyOS and nesC.
> 
> Sir, is this the right way of learning?

Start with the reference, tos-programming.  Then try to work through the basic tutorials.  http://tinyos.stanford.edu/tinyos-wiki/index.php/TinyOS_Tutorials

> Sir, as you said, tinyOS and nesC combination is harder, is there any other language for TinyOS?

nesC is the implementation language.

cire831, 2018-12-14:
anita, these kinds of questions are more appropriate for tinyos-help (tinyos-help@millennium.berkeley.edu)

anitaSpatil, 2018-12-15:
Tanks a lot Sir.

anitaSpatil, 2018-12-15:
Thanks a lot Sir.