tinyos/tinyos-main #429: To Update GPG Signature for Ubuntu18.04

inouetoukyou, 2018-12-12:
With the method in http://tinyprod.net/repos/debian to install tinyos in Ubuntu18.04, it will enter an error "signatures were invalid".
Though we can force installing by appending arguments `-o APT::Get::AllowUnauthenticated=true`, it is better to update the signature.

anitaSpatil, 2018-12-12:
First time I have installed on Ubuntu 14.04, later when my Ubuntu got upgraded to 16, tinyos stopped working, without knowing any reason I kept on trying to install on 16 and 18 also, even I changed their sources.list file also. Finally I came back to Ubuntu 14.04 LTS, it got successfully installed. Now I have tinyos-2.1.2 on Ubuntu 14.04.

inouetoukyou, 2018-12-12:
I agree with you, but I prefer a better experience in ubuntu18

anitaSpatil, 2018-12-13:
Thank you for suggestion. I will try , if I get stuck while upgrading signature I will contact again.
May I know what sort of work you are doing on TinyOS?

inouetoukyou, 2018-12-13:
I am a postgraduate student. Recently, I am struggling with writing an app to read noise floor(RSSI).

anitaSpatil, 2018-12-13:
I am basically Computer Science and Engineering student, not having much knowledge about electronic side. Now I am the research scholar  in "Operating Systems for Wireless Sensor Networks " domain. 
In which language you are developing the app?
this link may help you
http://tinyos.stanford.edu/tinyos-wiki/index.php/Rssi_Demo

inouetoukyou, 2018-12-13:
Thanks. For more information, I want to know which SNS App do you prefer, after all, talking too much in this site can annoys other people.

cire831, 2018-12-14:
> Sorry, if anybody got annoyed because of my conversation. I am new in the
> group and thought, the discussions will be with respect to particular
> issues and only interested people get involved in it by notifications. With
> the hope of getting some support I gave my details.
> Thank you all in the group.

not a problem but more appropriate for tinyos-help,

tinyos-help@millennium.berkeley.edu

giovannirosa, 2019-01-06:
I'm a graduate student in Computer Science and also interested in using tinyos in Ubuntu 18.04. Currently I installed manually from git repos.

cire831, 2019-05-04:
actively working on it.

cire831, 2019-05-23:
done.   the tinyprod.net repository has been rebuilt and signed with a new key.  you will have to reimport the tinyprod key from the web site.

closed 2019-05-23 by cire831