tinyos/tinyos-main #432: Issue building to TOSSIM

clegaard, 2019-02-22:
Hi,

I am currently doing an university course in wireless sensor networks using tinyOS. 
Building to actual hardware platforms works (telosb & micaz), however i am having trouble building to TOSSIM.

When i try to build the Blink example i get the following output:

<pre><font color="#8AE234"><b>clegaard@ubuntu</b></font>:<font color="#729FCF"><b>~/Desktop/tinyos-main/apps/Blink</b></font>$ make GCC=gcc-4.8 GPP=g++-4.8 micaz sim
<font color="#729FCF"><b>[INFO]</b></font> placing object files in simbuild/micaz
<font color="#729FCF"><b>[INFO]</b></font> writing XML schema to app.xml
<font color="#729FCF"><b>[INFO]</b></font> compiling BlinkAppC to object file sim.o
nescc -c -shared -fPIC -o simbuild/micaz/sim.o -g -O0 -gcc=gcc-4.8 -Wnesc-all -fnesc-include=tos -fnesc-scheduler=TinySchedulerC,TinySchedulerC.TaskBasic,TaskBasic,TaskBasic,runTask,postTask -fnesc-cfile=simbuild/micaz/app.c -fnesc-separator=__ -DTOSSIM -fnesc-nido-tosnodes=1000 -fnesc-simulate -fnesc-nido-motenumber=sim_node\(\) -fnesc-gcc=gcc-4.8 -I../../tos/lib/tossim -I../../tos/platforms/micaz/sim -I../../tos/platforms/micaz -I../../tos/platforms/mica/sim -I../../tos/platforms/mica -I../../tos/platforms/micaz/chips/cc2420/sim -I../../tos/platforms/micaz/chips/cc2420 -I../../tos/chips/cc2420/sim -I../../tos/chips/cc2420 -I../../tos/chips/cc2420/alarm -I../../tos/chips/cc2420/control/sim -I../../tos/chips/cc2420/control -I../../tos/chips/cc2420/csma/sim -I../../tos/chips/cc2420/csma -I../../tos/chips/cc2420/interfaces/sim -I../../tos/chips/cc2420/interfaces -I../../tos/chips/cc2420/link/sim -I../../tos/chips/cc2420/link -I../../tos/chips/cc2420/lowpan/sim -I../../tos/chips/cc2420/lowpan -I../../tos/chips/cc2420/lpl -I../../tos/chips/cc2420/packet/sim -I../../tos/chips/cc2420/packet -I../../tos/chips/cc2420/receive/sim -I../../tos/chips/cc2420/receive -I../../tos/chips/cc2420/spi -I../../tos/chips/cc2420/transmit -I../../tos/chips/cc2420/unique/sim -I../../tos/chips/cc2420/unique -I../../tos/platforms/mica2/chips/at45db -I../../tos/platforms/mica/chips/at45db -I../../tos/chips/at45db -I../../tos/chips/atm128/sim -I../../tos/chips/atm128 -I../../tos/chips/atm128/adc -I../../tos/chips/atm128/pins/sim -I../../tos/chips/atm128/pins -I../../tos/chips/atm128/spi/sim -I../../tos/chips/atm128/spi -I../../tos/chips/atm128/timer/sim -I../../tos/chips/atm128/timer -I../../tos/lib/power -I../../tos/lib/timer -I../../tos/lib/serial -fnesc-no-debug -DPLATFORM_MICAZ -Wall -Wshadow -finline-limit=100000 -I../../tos/system -I../../tos/types -I../../tos/interfaces -DIDENT_APPNAME=\&quot;BlinkAppC\&quot; -DIDENT_USERNAME=\&quot;clegaard\&quot; -DIDENT_HOSTNAME=\&quot;ubuntu\&quot; -DIDENT_USERHASH=0xcdb9a605L -DIDENT_TIMESTAMP=0x5c705794L -DIDENT_UIDHASH=0x00cba8feL -Wno-nesc-data-race BlinkAppC.nc   -fnesc-dump=components -fnesc-dump=variables -fnesc-dump=constants -fnesc-dump=typedefs -fnesc-dump=interfacedefs -fnesc-dump=tags -fnesc-dumpfile=app.xml
In file included from /usr/include/stdlib.h:55,
                 from ../../tos/lib/tossim/tos.h:59:
/usr/include/x86_64-linux-gnu/bits/floatn.h:75: syntax error before `float&apos;
/usr/include/x86_64-linux-gnu/bits/floatn.h:87: syntax error before `_Float128&apos;
In file included from /usr/include/math.h:420,
                 from ../../tos/lib/tossim/tos.h:60:
/usr/include/x86_64-linux-gnu/bits/mathcalls-helper-functions.h:21: syntax error before `__value&apos;
/usr/include/x86_64-linux-gnu/bits/mathcalls-helper-functions.h:25: syntax error before `__value&apos;
/usr/include/x86_64-linux-gnu/bits/mathcalls-helper-functions.h:30: syntax error before `__value&apos;
/usr/include/x86_64-linux-gnu/bits/mathcalls-helper-functions.h:33: syntax error before `__value&apos;
/usr/include/x86_64-linux-gnu/bits/mathcalls-helper-functions.h:36: syntax error before `__value&apos;
/usr/include/x86_64-linux-gnu/bits/mathcalls-helper-functions.h:39: syntax error before `__x&apos;
/usr/include/x86_64-linux-gnu/bits/mathcalls-helper-functions.h:42: syntax error before `__value&apos;
../../support/make/extras/sim.extra:67: recipe for target &apos;sim-exe&apos; failed
make: *** [sim-exe] Error 1
</pre>

I realize that all of you have limited resources, however any help or pointers are greatly appreciated.

cire831, 2019-02-22:
> *clegaard@ubuntu*:*~/Desktop/tinyos-main/apps/Blink*$ make GCC=gcc-4.8 GPP=g++-4.8 micaz sim*[INFO]* placing object files in simbuild/micaz*[INFO]* writing XML schema to app.xml*[INFO]* compiling BlinkAppC to object file sim.o
> nescc -c -shared -fPIC -o simbuild/micaz/sim.o -g -O0 -gcc=gcc-4.8 -Wnesc-all -fnesc-include=tos -fnesc-scheduler=TinySchedulerC,TinySchedulerC.TaskBasic,TaskBasic,TaskBasic,runTask,postTask -fnesc-cfile=simbuild/micaz/app.c -fnesc-separator=__ -DTOSSIM -fnesc-nido-tosnodes=1000 -fnesc-simulate -fnesc-nido-motenumber=sim_node\(\) -fnesc-gcc=gcc-4.8 -I../../tos/lib/tossim -I../../tos/platforms/micaz/sim -I../../tos/platforms/micaz -I../../tos/platforms/mica/sim -I../../tos/platforms/mica -I../../tos/platforms/micaz/chips/cc2420/sim -I../../tos/platforms/micaz/chips/cc2420 -I../../tos/chips/cc2420/sim -I../../tos/chips/cc2420 -I../../tos/chips/cc2420/alarm -I../../tos/chips/cc2420/control/sim -I../../tos/chips/cc2420/control -I../../tos/chips/cc2420/csma/sim -I../../tos/chips/cc2420/csma -I../../tos/chips/cc2420/interfaces/sim -I../../tos/chips/cc2420/interfaces -I../../tos/chips/cc2420/link/sim -I../../tos/chips/cc2420/link -I../../tos/chips/cc2420/lowpan/sim -I../../tos/chips/cc2420/lowpan -I../../tos/chips/cc2420/lpl -I../../tos/chips/cc2420/packet/sim -I../../tos/chips/cc2420/packet -I../../tos/chips/cc2420/receive/sim -I../../tos/chips/cc2420/receive -I../../tos/chips/cc2420/spi -I../../tos/chips/cc2420/transmit -I../../tos/chips/cc2420/unique/sim -I../../tos/chips/cc2420/unique -I../../tos/platforms/mica2/chips/at45db -I../../tos/platforms/mica/chips/at45db -I../../tos/chips/at45db -I../../tos/chips/atm128/sim -I../../tos/chips/atm128 -I../../to [... cut ...] 705794L -DIDENT_UIDHASH=0x00cba8feL -Wno-nesc-data-race BlinkAppC.nc   -fnesc-dump=components -fnesc-dump=variables -fnesc-dump=constants -fnesc-dump=typedefs -fnesc-dump=interfacedefs -fnesc-dump=tags -fnesc-dumpfile=app.xml
> ../../support/make/extras/sim.extra:67: recipe for target 'sim-exe' failed

looks like it is mixing tool chains.   the reference to /usr/include/x86_64
seems to indicate pulling in the default install on the system but you are
specifing 4.8 on the command line.

the last time this stuff was built it was done with a much older tool chain.

try removing the GCC= and GPP= parts and see if that works.

I've never used the simulator.

giovannirosa, 2019-04-04:
That was happening to me when I tried to install and use TinyOS in modern Ubuntu releases.

That's definitely something with gcc version.

Try to use Ubuntu 14.04, it worked for me.

cire831, 2019-04-05:
I'm in the process of updating the toolchains for msp430, msp432, etc.   Updating to a new base of Ubuntu 16.04 but will also be tested on 18.04.  This will also work with Debian Stretch (9)

garyquiet, 2019-12-10:
Hi, @clegaard @sadatd . I met the same problem. Have you solved this issue? If yes, would you please share the solution to me. Thanks.

sadatd, 2019-12-10:
Hello @garyquiet . I could't make it work normally. But together with @ileri  we manage to make it work in docker. Here is link [https://hub.docker.com/r/serhatcileri/tinyos-ready](url) for tiny-os with java preinstalled.

clegaard, 2019-12-11:
Hi, @garyquiet a solution is to use a older build toolchain. Since i was using a virtual machine i simply downloaded an older distribution of Ubuntu, 15 as far as i remember.

Another solution may be installing an older version build chain on you machine. However, i don't have any experience with installing multiple of these on the same machine.

garyquiet, 2019-12-11:
> Hi, @garyquiet a solution is to use a older build toolchain. Since i was using a virtual machine i simply downloaded an older distribution of Ubuntu, 15 as far as i remember.
>
> Another solution may [...]

Thank you for your reply. I use the ubuntu 18.04.3 and tinyos's version is 2.1.2. As you said, i should change the ubuntu 18.04.3 to 15 or older than 15, right?

cire831, 2019-12-13:
no progress however couple of workarounds are listed above.

closed 2019-12-13 by cire831