tinyos/tinyos-main #433: problem when installing tinyos

jabberimaher, 2019-05-20:
HI,
please i try to install tinyos under ubuntu, i change sources.list by adding

tinyOS Repository
deb http://tinyos.stanford.edu/tinyos/dists/ubuntu lucid main
i save and i launch sudo apt-get update
it dispaly
E: The repository 'http://tinyos.stanford.edu/tinyos/dists/ubuntu lucid Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by defaul
when i launch sudo apt-get install- 2.1.2
it display:
Reading package lists... Done
Building dependency tree 
Reading state information... Done
E: Unable to locate package tinyos-2.0.2
E: Couldn't find any package by glob 'tinyos-2.1.2'
E: Couldn't find any package by regex 'tinyos-2.1.2'

cire831, 2019-05-23:
you want to use the instructions in https://github.com/tinyos/tinyos-main/blob/master/doc/00c_Setting_Up_Debian_Development.

it will point you at the tinyprod.net repository.

closed 2019-05-23 by cire831

jabberimaher, 2019-05-26:
hi
when i try to install packages, there is errors:
root@maher-Lenovo-G50-80:/home/maher# apt-get install openssh-client openssh-server
Reading package lists... Done
Building dependency tree       
Reading state information... Done
openssh-client is already the newest version (1:7.6p1-4ubuntu0.3).
openssh-server is already the newest version (1:7.6p1-4ubuntu0.3).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 msp430-46 : Depends: msp430-binutils-46 but it is not going to be installed
             Depends: msp430-gcc-46 but it is not going to be installed
             Depends: msp430-libc-46 but it is not going to be installed
             Depends: msp430mcu-46 but it is not going to be installed
             Depends: tinyos-tools but it is not going to be installed or
                      tinyos-tools-devel but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
when i launch apt --fix-broken install:
root@maher-Lenovo-G50-80:/home/maher# apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages were automatically installed and are no longer required:
  libde265-0 libftdi1 libgssdp-1.0-3 libgupnp-1.0-4 libgupnp-igd-1.0-4
  libjama-java liblilv-0-0 libserd-0-0 libsord-0-0 libsratom-0-0 libssl-doc
  libvo-aacenc0 libvo-amrwbenc0 nescc uisp
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  msp430-binutils-46 msp430-gcc-46 msp430-libc-46 msp430mcu-46 nesc
  tinyos-tools
The following NEW packages will be installed:
  msp430-binutils-46 msp430-gcc-46 msp430-libc-46 msp430mcu-46 nesc
  tinyos-tools
0 upgraded, 6 newly installed, 0 to remove and 364 not upgraded.
14 not fully installed or removed.
Need to get 0 B/74.1 MB of archives.
After this operation, 880 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 222833 files and directories currently installed.)
Preparing to unpack .../0-msp430-binutils-46_2.21.1-LTS20120406-tinyprod_amd64.deb ...
Unpacking msp430-binutils-46 (2.21.1-LTS20120406-tinyprod) ...
dpkg: error processing archive /tmp/apt-dpkg-install-lvzLMo/0-msp430-binutils-46_2.21.1-LTS20120406-tinyprod_amd64.deb (--unpack):
 trying to overwrite '/usr/msp430/lib/ldscripts/msp430.xbn', which is also in package binutils-msp430 2.22~msp20120406-5.1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../1-msp430-gcc-46_4.6.3-LTS20120406+20120911-tinyprod_amd64.deb ...
Unpacking msp430-gcc-46 (4.6.3-LTS20120406+20120911-tinyprod) ...
dpkg: error processing archive /tmp/apt-dpkg-install-lvzLMo/1-msp430-gcc-46_4.6.3-LTS20120406+20120911-tinyprod_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/gcc/msp430/4.6.3/include-fixed/syslimits.h', which is also in package gcc-msp430 4.6.3~mspgcc-20120406-7ubuntu5
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../2-msp430-libc-46_20120224-LTS20120406+20120502-tinyprod_amd64.deb ...
.........
may be i need to re-install ubuntu (i use 18.04 version)
best regards