tinyos/tinyos-main #437: Compile msp430-gcc

inouetoukyou, 2019-10-14:
Is there a tutorial about how to compile msp430-gcc?
I just installed tinyos in Mac OS X successfully by coping
the msp430-gcc binary, whose version, however is 3.4.
I want msp430-gcc with version 4.6, so it will work with
the neweset tinyos(2.1.2 or above).

cire831, 2019-10-14:
there was but I don't remember where it was done.

given the architecture of the msp430 cpus it is problematic and generally a pain in the tush.  We used to use the msp4305438 cpu but transistioned over to ARM based msp430p401r cpus.

cire831, 2019-10-14:
The ARM processors are fully supported over on the current master repository at https://github.com/tp-freeforall/prod

closed 2019-10-14 by cire831