tinyos/tinyos-main #439: Dynamic Linking in TinyOS

ShishirPatil, 2020-05-27:
There have been quite a few papers on dynamic linking for TinyOS ([1](https://www.ds.informatik.uni-kiel.de/en/publications/papers/2010-05-munawar-icc-dynamictinyos.pdf),[2](https://dl.acm.org/doi/10.1145/1644038.1644052)). 

1. I just wanted to check if the ability to do dynamic linking has been mainstreamed (merged to master)? If not, may I understand the challenges and issues. 
2. If dynamic linking has been mainstreamed, would this also work on MSP430, Cortex M0, and Cortex M4 class devices, or are there any limitations?

Please forgive me if this question is answered elsewhere.

cire831, 2020-05-27:
no problem, feel free.

reference 1 talks about dynamic module loading.  what problem are you trying to solve with this?

tinyos  systems clearly are orientated to Wireless Sensor networks with very limited resources.  So there is some kind of radio that is used to download a new module.  But once again what problem are you trying to solve and is it worth the additonal complexity.

Further with modern small embeded cpus (such as arm m0, m4 etc.), one can download the module but where does one put it (RAM?), flash, etc.  and how does one resolve symbol bindings etc so one can actually get to the interface.

further, one of the primary design goals of tinyos was to mitigate many of the problem techniques  that lead to code that is hard to make correct and is hard to maintain.  Dynamic loading and/or linking complicates the dynamic behaviour and would be much harder to maintain.

Another attibute of tinyos nesc programs is whole program compilation and optimization.  this is a good thing.  how does this work with dynamic loading?

back to the problem you are trying to solve.    What is it, how would dynamic loading solve this problem, and is it worth the complexity?

cire831, 2020-05-27:
reference 2 refers to a paper written about TOSThreads ...   TOSThreads was fully implemented at one point.   It was implemented to provide infrastructure for long running background computations or similar problems.

However, no long running computation code was ever implemented and over time the code no longer worked properly.  Nothing is free and there was significant impact to all interrupt handlers.

so TOSThreads was deprecated a while ago.

what does TOSThread and dynamic loading have to do with each other?

closed 2020-05-28 by cire831