tinyos/tinyos-main #441: test

arvincubus2, 2020-09-26:
test

arvincubus2, 2020-09-26:
package com.arvin;

import java.util.Scanner;
import java.text.NumberFormat;
import java.util.Locale;

public class Calculator {
double input1, input2, total;

 public void Calculate(){
 Scanner scan = new Scanner(System.in);
 System.out.println("CALCULATOR APPLICATION");
 System.out.print("\nEnter 1st Number: ");
 
 
 String d1 = scan.next();
 
 try{
 
 double input1 = Double.parseDouble(d1);
 this.input1 = input1;
 
 } catch (IllegalArgumentException e){
 System.out.println("Wrong Input" );
 }
 
 System.out.print("Enter 2nd Number: ");
 String d2 = scan.next();
 scan.close();
 
 try {
 
 double input2 = Double.parseDouble(d2);
 this.input2 = input2;
 
 } catch (NumberFormatException e){
   System.out.println("Wrong Input");
 }
 
 if (input1 >= 0 && input2 >= 0){
   total = input1 + input2;
   double newTotal = total;
   
   NumberFormat numFormat =     NumberFormat.getInstance(new Locale("en", "PH"));
   String formattedTotal = numFormat.format(newTotal);
   System.out.println("\nTotal: " + formattedTotal);
 }

 
 }
}

cire831, 2020-09-27:
please don't do this.

test in your own repository.  thanks.

closed 2020-09-27 by cire831