tinyos/tinyos-main #442: What should I do if I can't connect to http://tinyprod.net/ when installing tinyos

RiversJin, 2020-12-03:
I am a beginner and am trying to experiment with tinyos.
This website was still available a few days ago, but when I tried to install avr-gcc yesterday, I found that this website seems to be unable to open.
Are there any other alternatives?

cire831, 2020-12-05:
you poke me.  I'll look into it.  Seems to be off line.

by the way, you might want to use tp-freeforall: https://github.com/tp-freeforall/prod instead of tinyos-main.

RiversJin, 2020-12-05:
> you poke me. I'll look into it. Seems to be off line.
> 
> by the way, you might want to use tp-freeforall: https://github.com/tp-freeforall/prod instead of tinyos-main.

Thank you very much!!  I have used the tp-freeforall to configure the developing environment.

closed 2020-12-05 by RiversJin